ES-EDU-SIEGMA/Embedded-Controller #58: Improve Unit Test Coverage

YULEMING, 2023-12-05:
* [ ] Repair Unit Test. Fix bugs and build it successfully.
* [ ] Figure out the current coverage of Unit Test.
* [ ] Extent the coverage of Unit Test.

YULEMING, 2023-12-11:
#### Repair Unit Test. Fix bugs and build it successfully.
* After fixing some naming and declaration errors, the unit-test can be built successfully now.
#### Figure out the current coverage of Unit Test.
It tests:
* dispenser
  * the structure of the dispenser (working)
  * the cycle of the dispenser (not working)

YULEMING, 2023-12-11:
### Extent the coverage of Unit Test.
Tasks list, from bottom to top, easy to hard:
* [ ] Limit switch
* [ ] Hardware helper
* [ ] serial UART
* [ ] tmc2209
* [ ] motor
* [ ] dispenser
* [ ] Rondell
* [ ] controller
* [ ] rondell controller

YULEMING, 2023-12-17:
Yes, just pushed [here](https://github.com/ES-EDU-SIEGMA/Embedded-Controller/tree/%2358-fix-improve-unit-test). But it's not finished yet. The CMakeLists.txt will keep changing and may not be able to be built directly.

DavidFederl, 2024-02-13:
Please see #81

closed 2024-02-19 by YULEMING