ES-EDU-SIEGMA/Embedded-Controller #71: Assemble new PCBs

DavidFederl, 2024-01-09:


DavidFederl, 2024-01-22:
@YULEMING and @HuangQing23: I tested your PCBs -> one of them has faulty TVS-Diodes (I fixed them)

---

You only need to add the female headers for the TMC/Tiny modules. I would recommend using a broken TMC/Tiny to hold them in place while soldering.

YULEMING, 2024-02-02:
* With the new PCB board, it stops in the while loop:
```c
void setUpMotor(Motor_t *motor, SerialAddress_t address) {

    TMC2209_setupByMotor(&motor->tmc2209,address);

    while (!TMC2209_isSetupAndCommunicating(&motor->tmc2209)) {
        if (TMC2209_disabledByInputPin(&motor->tmc2209)) {
            PRINT_DEBUG("Setup: Stepper driver with address %u DISABLED by input pin!", address)
        }
        PRINT_DEBUG("Setup: Stepper driver with address %u NOT communicating and setup!", address)
        TMC2209_setupByMotor(&motor->tmc2209,address);
        sleep_ms(500);
    }
    ...
}
```
with this output:
`[MOTOR: setUpMotor] Setup: Stepper driver with address 0 NOT communicating and setup!.`
**The problem is there is something wrong when writing or reading register `GlobalConfig`**
**The `pdn_disable` bit is 1 in the message we write into the register, but is 0 when we read.**
* The condition of the while loop is checking register `GlobalConfig`:
```c
const static uint8_t ADDRESS_GCONF = 0x00;
typedef union GlobalConfig {
    struct {
        ...
        uint32_t pdn_disable : 1;
        ...
    };
    uint32_t bytes;
} TMC2209_GlobalConfig_t;
```
the `pdn_disable` bit. The while loop is done when this bit is `1`.
* We write this bit here:
```c
void setUpMotor(Motor_t *motor, SerialAddress_t address) {

    TMC2209_setupByMotor(&motor->tmc2209,address);

    ...
}
```
And we check `pdn_disable` right after `TMC2209_setupByMotor(...);`:
```c
void setUpMotor(Motor_t *motor, SerialAddress_t address) {

    TMC2209_setupByMotor(&motor->tmc2209,address);

    while (!TMC2209_isSetupAndCommunicating(&motor->tmc2209)) {
        ...
    }
    ...
}
```
* If `pdn_disable` is not `1`, we setup it again:
```c
void setUpMotor(Motor_t *motor, SerialAddress_t address) {

    TMC2209_setupByMotor(&motor->tmc2209,address);

    while (!TMC2209_isSetupAndCommunicating(&motor->tmc2209)) {
        ...
        TMC2209_setupByMotor(&motor->tmc2209,address);
        ...
    }
    ...
}
```
* The tmc setup `GlobalConfig` by calling `TMC2209_setOperationModeToSerial` inside `TMC2209_setup`:
```c
void TMC2209_setOperationModeToSerial(TMC2209_t *tmc2209, serialUart_t serial,
                                      long serial_baud_rate, SerialAddress_t serial_address) {
    ...
    tmc2209->global_config.pdn_disable = 1;
    ...

    TMC2209_writeStoredGlobalConfig(tmc2209);
}
```
as you can see `pdn_disable` is 1.
* Here we read the value of `pdn_disable`
```c
bool TMC2209_serialOperationMode(TMC2209_t *tmc2209) {
    TMC2209_GlobalConfig_t global_config;
    global_config.bytes = TMC2209_readGlobalConfigBytes(tmc2209);

    return global_config.pdn_disable;
}
```
the `global_config.pdn_disable` is returned again:
```c
bool TMC2209_isSetupAndCommunicating(TMC2209_t *tmc2209) {
    return TMC2209_serialOperationMode(tmc2209);
}
```
you can find it's 0 when you print the debug message.
All the `read` and `write` here are finally executed by `TMC2209_read` and `TMC2209_write`. If they worked well in the past, then there is nothing wrong with them?

* I tested the integration test of the limit switch and motor. When testing the limit switch, there is no change when I open or close the limit switch. When testing the motor, it also stops in the same while loop.

###Do you have any idea that anything might influence this?
* The new design of the PCB?
* The tmc is broken?
* The change of the enable pin in the code?
* Or something is wrong with the code, but somehow they still worked in the past?

YULEMING, 2024-02-02:
The PCB board with `1A` and `2A` labels, when testing the limit switch, no change when I close and open the limit switch.
The PCB board on the machine, all limit switches work well. The state of limit switch changed when I open and close them.
The rest PCB board, limit switch 1 and 2 work well. The limit switch 3 no state change. When I close and open limit switch 4, the states of limit switch 3 and 4 change together.
But all TMCs are not working on all 3 PCB boards.

DavidFederl, 2024-02-05:
The limitswitches are working as expected now. The TMC2209 are still not reachable. I'll try to figure it out, by checking the UART communication.

DavidFederl, 2024-02-06:
I would think the TMC2209 requires some form of adjustment to work properly. -> [FAQ for Setup](https://learn.watterott.com/de/silentstepstick/faq/#wie-stellt-man-den-motorstrom-ein)

DavidFederl, 2024-02-07:
> The limitswitches are working as expected now. The TMC2209 are still not reachable. I'll try to figure it out, by checking the UART communication.

The UART communication is working now. The resistor value was specified to large. It has to be 270Ω.

closed 2024-02-07 by DavidFederl