ES-EDU-SIEGMA/Embedded-Controller #80: adjust TMC2209 to perform correctly

DavidFederl, 2024-02-07:


YULEMING, 2024-02-07:
[image]

So the `A` and `B` of the motor are like this, but I don't know which one is A or B on the motor. 
But the motor 04 on the `Left` is working well, so I moved it's connector from `M4` to `M1` and it also worked well.
The `M4` is closer to the TINY than `M1`, so I moved the connector of motor 01 from `M1` to `M4` but it still didn't work well. In this case, the cables' order is the same as motor 04.
Then I changed the order of the colored cables but none of them worked as well as motor 04.

I also exchanged the position of TMC2209 of `M4` and `M1`, but it also didn't work.

On the TMC2209 there are A1 and A2, I'm also not sure which one is A.

Might this have anything to do with the problem now on TMC2209?

DavidFederl, 2024-02-07:
The A B coils of the motor are provided with a current through the TMC2209. This current is generated by a PWM so that the motor is running smoothly. 
To my point of view the TMC2209 needs to be adjusted so that the output current is set correct, like mentioned [in this guide](https://learn.watterott.com/de/silentstepstick/faq/#wie-stellt-man-den-motorstrom-ein). I would checking this by comparing a TMC2209 that is working correctly and comparing it to one not working.

DavidFederl, 2024-02-09:
Measure the current between GND and V_Ref Pin on a working TMC2209 and compare it to the non working TMC2209. If there is a difference adjust the Potentiometer to get the value measured on the working TMC2209.

DavidFederl, 2024-02-13:
# The hardware is now working as expected.

## Fixed Problems

- The tiny2040 were flashed with the wrong version of the integration test. => **Build from develop branch for now!**
- The Crimp contacts for one the motor cables failed which caused problems. => **Replaced the faulty parts**

closed 2024-02-13 by DavidFederl