ES-EDU-SIEGMA/Embedded-Controller #84: Fix the problem when testing left/rondell/right

HuangQing23, 2024-02-15:
When we use USB to connect three pcb boards to test left/rondell/right, it always stops at the execution to output  "moveDown" in 'PRINT_DEBUG("moveDown")' of 'resetDispenserPosition(dispenser_t *dispenser)'. Function 'moveMotorDown' was executed but it seems like the motors are not running.

YULEMING, 2024-02-19:
* Yes, and by integrationTest_Motor it calls the same function `moveMotorDown` but it works.
Due to it doesn't work in `resetDispenserPosition` `debug` `left`, it sticks in the following while loop:
```c
static void resetDispenserPosition(dispenser_t *dispenser) {
    ...
    moveMotorDown(dispenser->address);
    PRINT_DEBUG("moveDown")
    while (!limitSwitchIsClosed(dispenser->limitSwitch))
        ;
    ...
```
* So the question is: why `moveMotorDown` works in `integrationTest_motor` `debug`, but not in `resetDispenserPosition` `debug` `left`?
* I don't think `moveMotorUp` and `stopMotor` will either work in `resetDispenserPosition` `debug` `left`.

HuangQing23, 2024-02-20:
fix the problem by #85

HuangQing23, 2024-02-20:
To make the motors work  properly, we should get the rope wrapped clockwise around the motor's rotating shaft. This is because our function 'moveMotorDowm' will make the motor shaft rotates clockwise. If the rope is wrapped in the opposite direction of the motor shaft rotation due to rope length or over-rotation of the shaft, then the input command will rotate in the opposite direction of the actual rotation.

closed 2024-02-21 by HuangQing23